motioneye-project/motioneye #3080: google drive work around to get authorization key

cinnamonsweets, 2024-12-01:
I feel this comment should be pinned until the developers have time to get google drive sorted out.

https://github.com/motioneye-project/motioneye/issues/2625#issuecomment-2430719316

ArgiesDario, 2025-01-21:
I tried that workaround, but it no longer seems to work. However, I believe it can be easily accomplished using rclone.

For reference, you can check [this comment](https://github.com/motioneye-project/motioneye/issues/2625#issuecomment-2603399850).

JamBalaya56562, 2026-11-03:
The workaround is no longer needed: #3512 fixed obtaining the Google Drive key in motionEye itself (see #2625). It is in 0.45.0b1 on PyPI and will be in v0.45.0. Closing as completed.

closed 2026-11-03 by JamBalaya56562