motioneye-project/motioneye #3081: pi camera module v2 not detected

bleporini, 2024-12-03:
Hello, 

I followed the troubelshooting, but nothing sorts this out:

Camera connected to the camera port
```
$ uname -a
Linux cam 6.6.51+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.51-1+rpt3 (2024-10-08) aarch64 GNU/Linux
$ grep bcm2835-v4l2 /etc/modules
bcm2835-v4l2
$ sudo lsmod| grep  bcm2835_v4l2
bcm2835_v4l2           40960  0
bcm2835_mmal_vchiq     36864  3 bcm2835_codec,bcm2835_v4l2,bcm2835_isp
videobuf2_vmalloc      12288  1 bcm2835_v4l2
videobuf2_v4l2         32768  6 bcm2835_codec,bcm2835_unicam,bcm2835_v4l2,rpivid_hevc,v4l2_mem2mem,bcm2835_isp
videodev              303104  22 v4l2_async,bcm2835_codec,v4l2_fwnode,imx219,videobuf2_v4l2,bcm2835_unicam,bcm2835_v4l2,rpivid_hevc,v4l2_mem2mem,bcm2835_isp
videobuf2_common       69632  10 bcm2835_codec,videobuf2_vmalloc,videobuf2_dma_contig,videobuf2_v4l2,bcm2835_unicam,bcm2835_v4l2,rpivid_hevc,v4l2_mem2mem,videobuf2_memops,bcm2835_isp
$ grep gpu /boot/firmware/config.txt
gpu_mem=128
```
The module is a genuine one.
`rpicam-still -o cam.jpg` worked like others rpicam utilities.

I tried MotionEyeOS, but it always fail with `start4.elf is not compatible`, either with the `writeimage.sh` script or with the RPI image writing software. 

Any clue?

bleporini, 2024-12-04:
And here's what I can find in the log:
```
Dec 03 18:26:39 cam motion[4084]: [0:motion] [NTC] [ALL] conf_load: Processing thread 0 - config file /etc/motioneye/motion.conf
Dec 03 18:26:39 cam motion[4084]: [0:motion] [NTC] [ALL] config_camera: Processing camera config file camera-1.conf
Dec 03 18:26:39 cam motion[4084]: [0:motion] [NTC] [ALL] motion_startup: Logging to syslog
Dec 03 18:26:39 cam motion[4084]: [0:motion] [NTC] [ALL] motion_startup: Motion 4.7.0 Started
Dec 03 18:26:39 cam motion[4084]: [0:motion] [NTC] [ALL] motion_startup: Using default log type (ALL)
Dec 03 18:26:39 cam motion[4084]: [0:motion] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
Dec 03 18:26:39 cam motion[4084]: [1:ml1:Camera1] [ERR] [VID] v4l2_device_capability: Device does not support capturing.
Dec 03 18:26:39 cam motion[4084]: [1:ml1:Camera1] [ERR] [VID] vid_start: V4L2 device failed to open
Dec 03 18:26:39 cam motion[4084]: [1:ml1:Camera1] [WRN] [ALL] motion_init: Could not fetch initial image from camera
Dec 03 18:26:39 cam motion[4084]: [1:ml1:Camera1] [WRN] [ALL] motion_init: Motion continues using width and height from config file(s)
Dec 03 18:26:40 cam motion[4084]: [1:ml1:Camera1] [WRN] [ALL] mlp_retry: Retrying until successful connection with camera
Dec 03 18:26:40 cam motion[4084]: [1:ml1:Camera1] [ERR] [VID] v4l2_device_capability: Device does not support capturing.
Dec 03 18:26:40 cam motion[4084]: [1:ml1:Camera1] [ERR] [VID] vid_start: V4L2 device failed to open
Dec 03 18:26:50 cam motion[4084]: [1:ml1:Camera1] [WRN] [ALL] mlp_retry: Retrying until successful connection with camera
Dec 03 18:26:50 cam motion[4084]: [1:ml1:Camera1] [ERR] [VID] v4l2_device_capability: Device does not support capturing.
Dec 03 18:26:50 cam motion[4084]: [1:ml1:Camera1] [ERR] [VID] vid_start: V4L2 device failed to open
Dec 03 18:27:00 cam motion[4084]: [1:ml1:Camera1] [WRN] [ALL] mlp_retry: Retrying until successful connection with camera
Dec 03 18:27:00 cam motion[4084]: [1:ml1:Camera1] [ERR] [VID] v4l2_device_capability: Device does not support capturing.
Dec 03 18:27:00 cam motion[4084]: [1:ml1:Camera1] [ERR] [VID] vid_start: V4L2 device failed to open
```

Matthew1471, 2024-12-04:
https://github.com/motioneye-project/motioneye/issues/2812#issuecomment-2506840345

DifferentComputers, 2025-05-30:
I have essentially the same problem getting MotionEye to display an image from an Arducam IMX219.

I don't get the "start4.elf is not compatible" error, but in MotionEye all I get from the cam is a grey screen and "UNABLE TO OPEN VIDEO DEVICE".

`Linux Dash 6.12.25+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.12.25-1+rpt1 (2025-04-30) aarch64 GNU/Linux`

`sudo lsmod| grep  bcm2835_v4l2
bcm2835_v4l2           40960  0
bcm2835_mmal_vchiq     36864  3 bcm2835_codec,bcm2835_v4l2,bcm2835_isp
videobuf2_vmalloc      12288  2 gspca_main,bcm2835_v4l2
videobuf2_v4l2         32768  6 gspca_main,bcm2835_codec,bcm2835_unicam_legacy,bcm2835_v4l2,v4l2_mem2mem,bcm2835_isp
videodev              311296  12 v4l2_async,gspca_main,bcm2835_codec,v4l2_fwnode,gspca_zc3xx,bcm2835_unicam_legacy,imx219,videobuf2_v4l2,bcm2835_v4l2,v4l2_mem2mem,bcm2835_isp
videobuf2_common       73728  10 gspca_main,bcm2835_codec,bcm2835_unicam_legacy,videobuf2_vmalloc,videobuf2_dma_contig,videobuf2_v4l2,bcm2835_v4l2,v4l2_mem2mem,videobuf2_memops,bcm2835_isp
`