motioneye-project/motioneye #3083: 0.43.1b2 - Noise Level Rounding Issue

Matthew1471, 2024-12-05:
I have noticed when I set noise level to 2% when it gets read back again it changes to 1%. Setting the noise level specifically to 1% does cause a lower noise value.. so the 2% is applying, it's just rounding down to 1% in the GUI when reading the value back.

```
admin@Camera1:~ $ grep -i noise_level /etc/motioneye/camera-1.conf
noise_level 5
admin@Camera1:~ $ grep -i noise_level /etc/motioneye/camera-1.conf
noise_level 3
```

zagrim, 2024-12-07:
Yes indeed, so it seems to go. Noise level of 2% converts to `noise_level 5` in camera config which is read back as being 1%. Correspondingly, noise level of 3% converts to 8 in config but that is read back correctly as 3%. I'd suspect a rounding goof-up in the value conversion, or a off-by-one error, but it took a while to spot the anomaly.

The Python code to convert the UI slider (scale 0 to 100) to Motion config parameter value (scale 0 to 255) is this (`ui[]` being dictionary of UI config settings):
```max(1, int(round(int(ui['noise_level']) * 2.55)))```

And the conversion back from config to UI is this (`data[]` being dictionary of config file parameters):
```int(int(data['noise_level']) / 2.55```

The culprit here being the outer `int()` in the second formula, which does effectively does same as `floor()`, that is rounds down.

I did a quick testing changing that and it seemed to fix the issue. I'll do a PR in a while.

Matthew1471, 2024-12-08:
Amazing, thank you for looking into this and providing a pull request!

I'm slowly getting to grips with the code too and submitted my pull request to fix when remote cameras are unavailable.

MichaIng, 2024-12-08:
Also shouldn't `max(1, int(round(int(ui['noise_level']) * 2.55)))` be equivalent to `max(1, round(int(ui['noise_level']) * 2.55))`, i.e. `int()` around `round()` nonsense?

So just to understand the logic:
1. You type a float into the UI.
2. It is rounded down, cutting decimal places.
3. It is multiplied with 2.55
4. It is rounded mathematically correct to the nearest integer. So `.0` to `.499...` will be rounded down, including `.5` to `.999...` will be rounded up. So a 2% becomes 5.1 becomes 5. And a 3% becomes 7.65 becomes 8.

When reading back the value:
1. The config value is first rounded down ... unnecessary as it should be an integer already. If it is a float, then dividing that float by 2.55 would be actually better than rounding it first. However does not really matter, as once storing it via GUI, it is assured to be an integer already. But for code simplicity, I would actually remove the inner `int()`, or can this cause any issues? Is maybe the `int()` error output of completely invalid values better than the error produced by the multiplication attempt of an invalid value?
2. It is divided by 2.55.
3. It was rounded down. Hence the previously stored 5 becomes an 1.96 becomes a 1. And the previously stored 8 becomes a 3.14 becomes a 3. So Yeah, it totally makes sense that this is wrong when input value is rounded down, so that the config value needs to be rounded up. Using `round()` solves it, as it assures rounding is done into the same direction in both cases. An alternative would be to use `ceil()` in one direction and `floor()` or `int()` in the other. But rounding mathematically correctly is of course preferred.

So yeah, at least the outer `int()` on the input value can be removed as well, and the inner `int()`s in both cases could be removed, if `int()` does not do better error handling/output, compared to multiplying or dividing non-numeric values.

MichaIng, 2024-12-08:
```console
root@dietpi:~# python3 -c 'print("a" * 3.3)'
Traceback (most recent call last):
  File "<string>", line 1, in <module>
TypeError: can't multiply sequence by non-int of type 'float'
root@dietpi:~# python3 -c 'print("a" / 3.3)'
Traceback (most recent call last):
  File "<string>", line 1, in <module>
TypeError: unsupported operand type(s) for /: 'str' and 'float'
root@dietpi:~# python3 -c 'print(int("a") / 3.3)'
Traceback (most recent call last):
  File "<string>", line 1, in <module>
ValueError: invalid literal for int() with base 10: 'a'
```
I guess we can remove the inner `int()` as well.

zagrim, 2024-12-09:
@MichaIng You're absolutely correct on the duplicated `int` conversion, and seems like also the other one can be removed. What came to my mind was handling of missing value (say, empty config), but that seems to produce the same (odd) default value of 13% noise level both with and without having explicit conversion to number first. So as far as I'm concerned, your addition looks good :+1:

zagrim, 2024-12-09:
But now, well, _looks are deceivin'_ :laughing: 

```   ERROR: can't multiply sequence by non-int of type 'float'
Traceback (most recent call last):
  File "/mnt/project/github/me-py-venv/lib/python3.12/site-packages/tornado/web.py", line 1790, in _execute
    result = await result
             ^^^^^^^^^^^^
  File "/mnt/project/github/me-py-venv/lib/python3.12/site-packages/motioneye/handlers/config.py", line 77, in post
    await self.set_config(camera_id)
  File "/mnt/project/github/me-py-venv/lib/python3.12/site-packages/motioneye/handlers/config.py", line 341, in set_config
    await set_camera_config(int(key), cfg, check_finished)
  File "/mnt/project/github/me-py-venv/lib/python3.12/site-packages/motioneye/handlers/config.py", line 158, in set_camera_config
    local_config = config.motion_camera_ui_to_dict(ui_config, local_config)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/mnt/project/github/me-py-venv/lib/python3.12/site-packages/motioneye/config.py", line 918, in motion_camera_ui_to_dict
    'noise_level': max(1, round(ui['noise_level'] * 2.55)),
                                ~~~~~~~~~~~~~~~~~~^~~~~~
TypeError: can't multiply sequence by non-int of type 'float'
```

zagrim, 2024-12-09:
I'm just not quite sure in my morning brain what is the type of value from ui[] and don't have the time to start debugging that out. But looks to me there has been some good reason for doing some of those explicit conversions.