motioneye-project/motioneye #3088: Motion does not record images

bbillp, 2024-12-18:
MotionEye 0.43.1b2 Motion version 4.7.0   Loaded 12/16/2024
Raspberry Pi 3b with Rasbian 11  32 bit loaded fresh 12/16/2024, Raspberry Pi V2 Camera

MotionEye is displaying live video Ok but never captures  motion images even if a flash light is shined into the camera.  
Motion detection is enabled
work schedule is enabled
detection motion during working schedule

What am I missing ???

bbillp, 2024-12-18:
I am looking in /var/lib/motioneye/Camera1   and have used sudo find -name *.jpg with no motion images found.

bbillp, 2024-12-20:
Camera interval mode saves images so its just motion detection that fails to record images.  There is nothing in the log file.

JamBalaya56562, 2026-11-13:
As the live view works and interval pictures are saved, motion detection itself does not trigger. One known cause: with Frame Change Threshold at 0 %, Motion skips detection altogether; this is fixed on the dev branch (#3548), and until the next release a threshold above 0 % or Automatic Threshold Tuning avoids it. As there was no reply since 2024, I'm closing this; if it still happens on the current version, please open a new issue with motionEye's debug log while moving in front of the camera.

closed 2026-11-13 by JamBalaya56562