motioneye-project/motioneye #3103: Can't see cameras on DVR PC BOX.

gquerol, 2025-01-20:
Hello, how are you? First of all I wanted to congratulate you on the project! 

I have a PCBOX DVR with 4 cameras, which through VLC and/or zonemaider I can see the cameras, but not with motioneye. 

After so much fighting I found the URL that my DVR uses to share the streaming.

The same is:

rtsp://192.168.1.XXX:554/user=USER&password=PASS&channel=1&stream=0.sdp

In the motioneye configuration I created a network camera and tried pasting the complete URL, without completing username and pass and completing username and pass, removing it from the URL and the error I always receive is "451 client error". 

Could it be that the URL of motioneye create to consume the stream does it wrong?

Thanks!

gquerol, 2025-01-20:
[0:motion] [NTC] [ALL] motion_start_thread: ID de cámara: 1 es de camera-1.conf
[0:motion] [NTC] [ALL] motion_start_thread: ID de cámara: 1 Nombre de cámara: Camera1 Servicio: rtsp:
[0:motion] [NTC] [ALL] main: Esperando que los hilos terminen, pid: 7550
[1:ml1:Camera1] [NTC] [ALL] motion_init: Cámara 1 iniciada: detección de movimiento Habilitado
[1:ml1:Camera1] [NTC] [VID] vid_start: Abriendo Netcam RTSP
[1:ml1:Camera1] [DBG] [ALL] util_parms_parse: Análisis: keepalive = off,tolerant_check = on
[1:ml1:Camera1] [DBG] [ALL] util_parms_add: Analizado:> keepalive <> off <
[1:ml1:Camera1] [DBG] [ALL] util_parms_add: Analizado:> tolerant_check <> on <
[1:ml1:Camera1] [INF] [NET] netcam_rtsp_set_parms: Configurando norma stream.
[1:ml1:Camera1] [DBG] [ALL] util_parms_add: Analizado:> decoder <> NULL <
[1:ml1:Camera1] [INF] [NET] netcam_rtsp_set_path: Configurando rtsp a través de netcam
[1:ml1:Camera1] [INF] [NET] netcam_rtsp_set_options: norma: Configuración de rtsp / rtmp
[1:ml1:Camera1] [DBG] [ALL] util_parms_add: Analizado:> rtsp_transport <> tcp <
[1:ml1:Camera1] [INF] [NET] netcam_rtsp_set_options: norma: opción: keepalive = off
[1:ml1:Camera1] [INF] [NET] netcam_rtsp_set_options: norma: opción: tolerant_check = on
[1:ml1:Camera1] [INF] [NET] netcam_rtsp_set_options: norma: opción: rtsp_transport = tcp
[1:ml1:Camera1] [INF] [ENC] ffmpeg_avcodec_log: method DESCRIBE failed: 451 ERROR
[1:ml1:Camera1] [ERR] [NET] netcam_rtsp_open_context: norma: no se puede abrir la cámara ( Camera1): Server returned 4XX Client Error, but not one of 40{0,1,3,4}
[1:ml1:Camera1] [INF] [NET] netcam_rtsp_cleanup: norma: apagando la cámara de red.
[1:ml1:Camera1] [NTC] [NET] netcam_rtsp_cleanup: Resolución normal: cierre completo.
[1:ml1:Camera1] [ERR] [VID] vid_start: Netcam RTSP no pudo abrirse
[1:ml1:Camera1] [WRN] [ALL] motion_init: No se pudo recuperar la imagen inicial de la cámara 
[1:ml1:Camera1] [WRN] [ALL] motion_init: El movimiento continúa usando el ancho y la altura de los archivos de configuración
[1:ml1:Camera1] [NTC] [ALL] image_ring_resize: Cambiar el tamaño del búfer de captura previa a 1 elementos
[1:ml1:Camera1] [NTC] [ALL] image_ring_resize: Cambiar el tamaño del búfer de captura previa a 21 elementos
[1:ml1:Camera1] [WRN] [ALL] mlp_retry: Reintentando hasta una conexión exitosa con la cámara
[1:ml1:Camera1] [NTC] [VID] vid_start: Abriendo Netcam RTSP
[1:ml1:Camera1] [DBG] [ALL] util_parms_parse: Análisis: keepalive = off,tolerant_check = on
[1:ml1:Camera1] [DBG] [ALL] util_parms_add: Analizado:> keepalive <> off <
[1:ml1:Camera1] [DBG] [ALL] util_parms_add: Analizado:> tolerant_check <> on <
[1:ml1:Camera1] [INF] [NET] netcam_rtsp_set_parms: Configurando norma stream.
[1:ml1:Camera1] [DBG] [ALL] util_parms_add: Analizado:> decoder <> NULL <
[1:ml1:Camera1] [INF] [NET] netcam_rtsp_set_path: Configurando rtsp a través de netcam
[1:ml1:Camera1] [INF] [NET] netcam_rtsp_set_options: norma: Configuración de rtsp / rtmp
[1:ml1:Camera1] [DBG] [ALL] util_parms_add: Analizado:> rtsp_transport <> tcp <
[1:ml1:Camera1] [INF] [NET] netcam_rtsp_set_options: norma: opción: keepalive = off
[1:ml1:Camera1] [INF] [NET] netcam_rtsp_set_options: norma: opción: tolerant_check = on
[1:ml1:Camera1] [INF] [NET] netcam_rtsp_set_options: norma: opción: rtsp_transport = tcp
[1:ml1:Camera1] [INF] [ENC] ffmpeg_avcodec_log: method DESCRIBE failed: 451 ERROR
[1:ml1:Camera1] [ERR] [NET] netcam_rtsp_open_context: norma: no se puede abrir la cámara ( Camera1): Server returned 4XX Client Error, but not one of 40{0,1,3,4}
[1:ml1:Camera1] [INF] [NET] netcam_rtsp_cleanup: norma: apagando la cámara de red.
[1:ml1:Camera1] [NTC] [NET] netcam_rtsp_cleanup: Resolución normal: cierre completo.
[1:ml1:Camera1] [ERR] [VID] vid_start: Netcam RTSP no pudo abrirse

gquerol, 2025-01-23:
Hello! any idea? Thanks!

zagrim, 2025-04-05:
The URL you included in the description seems a bit weird. An URL with request parameters should have a question mark separating the parameters from the path (and not a slash, as shown in the URL you included), like this: 
`rtsp://192.168.1.XXX:554?user=USER&password=PASS&channel=1&stream=0.sdp`
(please note the question mark between "554" and "user")
However, your camera might as well use an URL which is not quite standard compliant. This is just to note a possible error in the URL, which also might be the reason it didn't work.

I would assume, that if you put the URL as-is (with the user and password included in the URL) AND left username and password empty in ME "Add camera" dialog, it should work.
The other reasonable combination (but this is just guessing) might be to use this URL:
`rtsp://192.168.1.XXX:554?channel=1&stream=0.sdp`
and the put the username and password in the corresponding fields in ME "Add camera" dialog.

JamBalaya56562, 2026-11-12:
zagrim's answer points at the likely cause: the parameters in that URL should follow a `?`, not a `/`, e.g. `rtsp://192.168.1.XXX:554?user=USER&password=PASS&channel=1&stream=0.sdp`. With the user and password in the URL, leave Username and Password empty in the Add Camera dialog. As there was no reply since, I'm closing this; please open a new issue with Motion's log if it still fails.

closed 2026-11-12 by JamBalaya56562