motioneye-project/motioneye #3104: Privacy Mask is only applied to video stream, but not to motion triggered movie

hubbty, 2025-01-21:
A Privacy Mask selected in the Video Device menu is only effective in video streams, but not in the motion triggered movies. To deserve its name, the mask should be effective in the recorded movies too.

Additinally I wonder, if a motion in the area of the mask does trigger recordings. I think it should not.

zagrim, 2025-01-23:
That sounds weird, since `mask_privacy` option in Motion (which is what the ME UI Privacy mask option ends up putting in the camera config) is supposed to not only mask the area from motion detection but also mask it from any viewing:
https://motion-project.github.io/motion_config.html#mask_privacy
(so basically both masks affect motion detection, and privacy mask affects also camera stream so you cannot even see what happens in that area of the frame).

Please check if your camera config really has `mask_privacy` in it.

hubbty, 2025-02-05:
Thank you! I had not edited the config file, but had used that menue on the left side to set the privacy mask. And the mask did show up while viewing the stream, but not in recorded videos. I had this mask set up in order to blacken the time overlay injected by the camera as the switching of the numbers had sometimes triggered recordings.

In the meantime I disabled the privacy mask and instead set up a new one in the Motion Settings menue. There were no recordings triggered by the time overlay of the camera since.

JamBalaya56562, 2026-11-12:
Thanks for reporting back. If Movie Passthrough is on under Movies, that explains it: Motion then writes the camera's own packets to the movie, and its docs say "text overlays, privacy masks etc will not be on the resulting video". Motion in a privacy masked area is not detected, as the privacy mask is applied before detection. For keeping the camera's clock out of detection, the detection mask you set up is the right tool. Closing as answered.

closed 2026-11-12 by JamBalaya56562