motioneye-project/motioneye #3106: Can't use cmd because keyboard isn't detected

DanielS833, 2025-01-25:
I have been trying to se up MotioneyeOS and have tried many different versions of it, it boots normally but whenever the login promt appears I can't type anything (it's not my keyboard because I have tested some other keyboards) I'm using raspberry pi 4 with 2gb RAM.
Also I have to add that I tried installing MotioneyeOS and always got the same error (start4.elf not compatible/start.elf not compatible), I fixed this by replacing the start.elf/start4.elf file with a different one I found online and it booted correctly.

DanielS833, 2025-01-25:
I managed to access the cmd line by connecting to the raspberry pi through SSH but running lsusb -v doesn't show any devices

zagrim, 2025-01-26:
You should ask this in MotionEyeOS repository: https://github.com/motioneye-project/motioneyeos/issues
There are likely more people who could help you with the OS level stuff.

This repository is solely for the MotionEye application.

closed 2025-01-27 by DanielS833