motioneye-project/motioneye #3116: Unable to choose specific resolution, padding gets added

Reperdix, 2025-02-17:
Hello!
I have been trying to set up motioneye on a Raspberry Pi 4 (4 GB ram) with a Raspberry Pi HQ camera. 
(Operating system: Linux raspberrypi 6.6.74+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.74-1+rpt1 (2025-01-27) aarch64
motionEye Version | 0.43.1b2
Motion Version | 4.7.0)

The camera only outputs these specific resolutions:
`    Modes: 'SRGGB10_CSI2P' : **1332x990** [120.05 fps - (696, 528)/2664x1980 crop]
           'SRGGB12_CSI2P' : **2028x1080** [50.03 fps - (0, 440)/4056x2160 crop]
                             **2028x152**0 [40.01 fps - (0, 0)/4056x3040 crop]
                             **4056x3040** [10.00 fps - (0, 0)/4056x3040 crop]`

And when motioneye tries to get some other resolution I get this error:
`[  131.417919] unicam fe801000.csi: Wrong width or height 1088x1920 (remote pad set to 4056x3040)
[  131.417938] unicam fe801000.csi: Failed to start media pipeline: -22`

The problem is that when I change the resolution in motioneye it does not change it to exactly what I wrote. 
This is from the motion.log in /var/log/motioneye
`[1:ml1:video0] [WRN] [VID] v4l2_pixfmt_stride: The image width(1080) is not multiple of the stride(2176)
[1:ml1:video0] [WRN] [VID] v4l2_pixfmt_stride: Image width will be padded 8 bytes
[1:ml1:video0] [WRN] [VID] v4l2_pixfmt_adj: Adjusting resolution from 1080x1920 to 1088x1920.`

It adds padding to the resolution. Sometimes it seems to want a multiple of 8, other times 16 or - as in this case - the stride(2176) (whatever that means).

Since the camera only seems to output specific resolutions I have to make MotionEye accept one of those. But I can't.

I would be very greatful for any help!

xandark, 2025-02-25:
Just a thought, but can you edit the actual config files motioneye is generating under the hood with the proper padded resolution that you need?

JamBalaya56562, 2026-11-14:
The `unicam … Wrong width or height` error comes from opening the sensor's raw device, which only takes the sensor's own modes. motionEye 0.45.0 can load libcamera's V4L2 layer for Motion (#3546): run `sudo /opt/motioneye/bin/motioneye_init --libcamera`, or set `libcamera_v4l2` in `/etc/motioneye/motioneye.conf`. libcamera then scales to the resolution you choose. On a Pi 5 with Camera Module 3 and Motion 4.7.1, 1920x1080, 2032x1520 and 4056x3040 all gave a picture this way; Motion still pads the width to the stride (2048 and 4096), which it only warns about. I could not test an HQ camera on a Pi 4, so please reopen if it still fails there with the layer.

closed 2026-11-14 by JamBalaya56562