motioneye-project/motioneye #3117: How to update to :main? under Debian 12

OberfragGER, 2025-02-17:
I have some warnings in my log and really want to change from to main.

I run Debian12 as a Proxmox LXC.

motionEye Version | 0.43.1b2
Motion Version | 4.7.0
OS Version | Linux 6.8.12-5-pve

I really want the Python 3.13 support from #3092 and dont have any clue how is it possible to update. I did the Upgrade Instruction but no sucess
```
sudo systemctl stop motioneye
sudo python3 -m pip install --upgrade --pre motioneye
sudo systemctl start motioneye
```

MichaIng, 2025-02-18:
Try
```sh
sudo python3 -m pip install --upgrade https://github.com/motioneye-project/motioneye
```
Looks like we need to do another beta release.

OberfragGER, 2025-02-18:
I will try that within the next days and will reply here.
Thanks a lot in advance.

Meanwhile i have finally found the docker-compose... The docker file seem to be the newest one, right?

closed 2026-02-08 by MichaIng