motioneye-project/motioneye #3137: RTSP camera not working

UKnau22, 2025-04-29:
I have MotionEye installed in HomeAssistant OS and got one rtsp camera working. I have a second identical camera that I added the same way, but that cam only shows a grey picture. The rtsp stream is accessible in Celluloid app on my Linux Mint laptop. Also, the HA built in generic camera stream works for this cam. I removed the cam several times, restarted the addon and HomeAssistant and then re-added the cam with the same result.
Any help is appreciated.

Here’s the system info:
Add-on version: 0.21.0
You are running the latest version of this add-on.
System: Home Assistant OS 15.2 (amd64 / qemux86-64)
Home Assistant Core: 2025.4.4
Home Assistant Supervisor: 2025.04.0

nikoss123, 2025-06-16:
Hi for the camera with grey picture try to change the video resolution for this camera (video device,video resolution,save) also check your rtsp link on vlc to ensure that your stream is working

JamBalaya56562, 2026-11-12:
There was no reply to the suggestions above since June 2025, so I'm closing this. If the second camera still shows a grey picture on the current add-on, please open a new issue with the add-on's log from the moment the camera is added.

closed 2026-11-12 by JamBalaya56562