motioneye-project/motioneye #3140: how to delete recorded video's on a raspberry pi in terminal

bob59, 2025-05-12:
Hi

I made a remote nestbox camera, using a Raspberry Pi 3 B running Bullseye with Motioneye, I was away for a few days and as the eggs hatched the birds were in/out a lot, I tried to access Motioneye on my local network but could not connect, I could only connect to the Raspberry Pi via puTTY, it looks like the card is full.

I found several folders containing video's at var/lib/motioneye/Camera1, but if I try to delete them using rm, I get an error 
rm: cannot remove '2025-05-06/13-48-37.mp4.thumb' :Read-only file system

could someone help please

thanks

saadqaz1, 2025-05-13:
Sounds like the pi put itself into RO mode to protect the storage. You might want to try running fsck or fdisk and then try removing the files. See https://askubuntu.com/questions/1322160/ubuntu-20-04-raspberry-pi-cant-write-anything-every-file-seems-to-be-in-read

MichaIng, 2025-05-13:
motionEye/motion can hog the camera device, when doing motion detection or while viewing the camera stream via web UI.

However, motion does not naively support the libcamera API. So if you have an RPi camera module attached and the "modern" camera API enabled, it won't be able to handle it with motionEye, without adding `libcamerafy`: #2425

What do you mean with "can't connect to motioneye"? You mean the web interface is not accessible? Check the service status:
```sh
sudo systemctl status motioneye
```

To test the camera module with `libcamera-hello`, it might be needed to stop the motionEye service first:
```sh
sudo systemctl stop motioneye
```

MichaIng, 2025-05-14:
Okay great.

In case it is not accessible again, check the full service logs:
```sh
journalctl -u motioneye
```

I'll mark this issue as closed then. Feel free to reopen if required.

closed 2025-05-14 by MichaIng