motioneye-project/motioneye #3141: Video Thumbnail Image Generation Issues

saadqaz1, 2025-05-13:
Hello,

I am running Motioneye in a Docker container on a Raspberrypi 4 streaming an IP camera on my local network. 

I sometimes have issues with the thumbnail of the videos being created and just get the default blank image like below:

[image]

I see these errors from the container:

```
   ERROR: failed to open media preview image file: cannot identify image file <StringIO.StringIO instance at 0xf442d558>

   ERROR: failed to create movie preview for /var/lib/motioneye/Camera1/2025-05-13/01-59-24.mp4: Command '[u'ffmpeg', u'-i', u'/var/lib/motioneye/Camera1/2025-05-13/01-59-24.mp4', u'-f', u'mjpeg', u'-vframes', u'1', u'-ss', u'4', u'-y', u'/var/lib/motioneye/Camera1/2025-05-13/01-59-24.mp4.thumb']' returned non-zero exit status 1
```

Has anyone faced a similar issue or have a work around? I have tried recreating the container/image but it still happens intermittently.

nasserbinlaboun, 2025-05-15:
Same issue, also I am running Motioneye in a Docker container

`   ERROR: failed to create movie preview for /var/lib/motioneye/Camera2/2025-05-15/20-02-45.mp4: Command '['ffmpeg', '-i', '/var/lib/motioneye/Camera2/2025-05-15/20-02-45.mp4', '-f', 'mjpeg', '-vframes', '1', '-ss', '4', '-y', '/var/lib/motioneye/Camera2/2025-05-15/20-02-45.mp4.thumb']' returned non-zero exit status 234.

   ERROR: failed to open media preview image file: cannot identify image file <_io.BytesIO object at 0x7f53cb429800>

   ERROR: failed to open media preview image file: cannot identify image file <_io.BytesIO object at 0x7f53c8c557b0>

   ERROR: failed to open media preview image file: cannot identify image file <_io.BytesIO object at 0x7f53c8c55760>

   ERROR: failed to open media preview image file: cannot identify image file <_io.BytesIO object at 0x7f53c8d56fc0>

   ERROR: failed to open media preview image file: cannot identify image file <_io.BytesIO object at 0x7f53c8c551c0>

   ERROR: failed to open media preview image file: cannot identify image file <_io.BytesIO object at 0x7f53c8c556c0>`

saadqaz1, 2025-05-17:
> Same issue, also I am running Motioneye in a Docker container
> 
> Version: edge `sha256:770d9868836f21ce75e095ec47a7ee90b5172c6ea3f37eee205890ce5a8f597a`
> 

Interestingly enough when I switched to `image: ghcr.io/motioneye-project/motioneye:edge@sha256:f8a08f37f069aa7fb419e60ffe73d998b798d8ae527be5992a8273ced48f1645` (the arm version mentioned in https://github.com/motioneye-project/motioneye/pkgs/container/motioneye ) it has been creating the video previews successfully and I also noticed a decent amount of reduction in cpu usage.

nasserbinlaboun, 2025-05-18:
Thank you, I tried this version `ghcr.io/motioneye-project/motioneye@sha256:3a38c03592203b6039873be026238a625da3f7bab099b93faf3024542dfae14c ` and it is working perfectly now.

MichaIng, 2025-11-23:
Are there probably some invalid image files, like null byte files generated from the earlier motionEye version which was affected by the issue? Since `ghcr.io/motioneye-project/motioneye@sha256:3a38c03592203b6039873be026238a625da3f7bab099b93faf3024542dfae14c` worked perfectly as you stated, all newer versions/builds should do as well.

I'll try to add the image path to the `failed to open media preview image file` error message, though I guess it is exactly the `/var/lib/motioneye/Camera2/2025-11-23/17-29-09.mp4.thumb` and others where `ffmpeg` filed before. Maybe some cleanup of the failed preview image is missing, or a graceful skip when trying to open non-existing preview files.

For the `ffmpeg` call, maybe we can catch the STDERR.

nasserbinlaboun, 2025-12-06:
@MichaIng 

I believe the issue is related to the file size of the videos being processed. It seems that:

- If the video file size is **less than 10MB**, no preview image is generated.
- For videos that are **10MB or larger**, a preview image is created.

[image]

MichaIng, 2026-03-05:
We have a working PR up 😃. But I need to check back with the author before I can/want to merge it, at best preserve proper credits. Can be tested regardless with `pip install --force-reinstall --no-deps` or applying the patch:
```sh
cd /usr/local/lib/python3*/dist-packages/motioneye
curl https://patch-diff.githubusercontent.com/raw/motioneye-project/motioneye/pull/3304.patch | sudo patch -p1
```