motioneye-project/motioneye #3144: RasperryPi with Docker Compose: ImportError: libcurl.so.4: cannot open shared object file: No such file or directory

alexwasserman, 2025-05-14:
I'm using the edge docker image in the docker-compose file:

```
services:
  motioneye:
    image: ghcr.io/motioneye-project/motioneye:edge
```

This is now stuck looping, unable to find `libcurl.so.4`:

```
motioneye@motion ~ % docker compose up
[+] Running 1/1
 ✔ Container motioneye-motioneye-1  Created                                                                                                                               0.0s
Attaching to motioneye-1
motioneye-1  | Traceback (most recent call last):
motioneye-1  |   File "/usr/local/bin/meyectl", line 5, in <module>
motioneye-1  |     from motioneye.meyectl import main
motioneye-1  |   File "/usr/local/lib/python3.13/dist-packages/motioneye/meyectl.py", line 29, in <module>
motioneye-1  |     from motioneye import config, settings
motioneye-1  |   File "/usr/local/lib/python3.13/dist-packages/motioneye/config.py", line 31, in <module>
motioneye-1  |     from motioneye import meyectl, motionctl, settings, tasks, uploadservices, utils
motioneye-1  |   File "/usr/local/lib/python3.13/dist-packages/motioneye/motionctl.py", line 29, in <module>
motioneye-1  |     from motioneye import mediafiles, settings, update, utils
motioneye-1  |   File "/usr/local/lib/python3.13/dist-packages/motioneye/mediafiles.py", line 39, in <module>
motioneye-1  |     from motioneye import config, settings, uploadservices, utils
motioneye-1  |   File "/usr/local/lib/python3.13/dist-packages/motioneye/uploadservices.py", line 32, in <module>
motioneye-1  |     import pycurl
motioneye-1  | ImportError: libcurl.so.4: cannot open shared object file: No such file or directory
```

MichaIng, 2025-05-14:
Hmm weird, can you access the container console and verify that `curl` and `libcurl4` are installed? `libcurl4t64` to be precise, but it contains `libcurl.so.4`.

... ah wait:
```
#25 1247.8 Removing libcurl4t64:riscv64 (8.13.0-1) ...
```
I guess it has to do with the back and forth with `curl` depending on GnuTLS flavor of `libcurl` in between, then switched back to OpenSSL flavor. Let me check latest builds.

MichaIng, 2025-05-14:
Yes, the latest build did not remove `libcurl4t64`. So it really was the dependency swap back and forth. Should not happen anymore now that Debian Trixie (base image for our container) is in hard freeze state.

Can you test it again with the latest Docker image, which has been updated just 7 minutes ago?