motioneye-project/motioneye #3151: Cached password for the admin user?

trharmon0, 2025-05-19:
I have been playing with motioneye as a docker container manged by Docker Compose and dockge. At one point I set the admin password on the web interface but since then I have:

- dockge Delete'd the container.
- removed the resulting dangling Docker volumes
- cleared the cookies in the browser

and yet when I respin the motioneye stack the new container and volumes are configured with the old admin password and even the previously configured USB video device.

Can you advise where else motioneye might be caching/storing the credentials and config?

zagrim, 2025-05-19:
MotionEye UI login passwords are saved to `/etc/motioneye/motion.conf` which you must not have deleted (perhaps it is mapped to somewhere else?) since otherwise your old admin password could possibly not work. The same applies to camera config, which is saved to `/etc/motioneye/camera-1.conf` (for the first created camera) and which is referenced by the main config file (`motion.conf`).

trharmon0, 2025-05-19:
Ty. That is useful guidance. I will take another look at the underlying volume mappings and it occurs to me that dockge might be doing a save on the underlying image also. Easy enough to check.

closed 2025-05-19 by trharmon0