motioneye-project/motioneye #3159: MotionEye and Ubuntu 25.04

SuMaPa, 2025-06-07:
Hello,

Next month, support for Ubuntu 24.10 ends. Before that happens, I'd like to upgrade to 25.04.

In principle, motioneye should be compatible, but has the 25.04 release been incorporated into motioneye's official support or packages? When I tried to install motioneye on 25.04 in May, I was unsuccessful.

Thank you very much.

SuMaPa, 2025-06-08:
I'm using the Ubuntu Minimal Server version (SSH only, no GUI), so it should actually work.
I assumed there would be a query to check if the distribution is supported or not.
If that's not the case, I can proceed with the update to 25.04.

Thank you very much.

MichaIng, 2025-06-08:
What can happen is that a new Python release breaks some old syntax/methods or modules motionEye depends on. It works well with currently latest Python 3.13.

You can also check whether the Ubuntu version you are heading for has a `motion` package: https://packages.ubuntu.com/motion
For V4L2 camera access and video conversion, additionally needed: `v4l-utils` `ffmpeg`
All no problem with current and upcoming Ubuntu versions, and not something which is gonna change all of a sudden, as both are well established tools.