motioneye-project/motioneye #3163: RTSP cameras only showing gret screen

starbasessd, 2025-06-17:
Setup:
Debian 12 (Server & Desktop XFCE4), 
Python 3.11.2, 
motionEye 0.43.1b4 (installed using instructions in this github)
motion version 4.7.0
4 ip cameras: 2 dlink mjpeg.cgi working, 2 foscam MPS2011
rtsp link: rtsp://<ip_address>:88/videoMain
foscams set to various resolutions, native, 800x600, 1280x720 (my default, works everywhere else, including older mE)
- - - - -
when set up, the 2 dlink cameras work great. the 2 foscams give the grey screen. All 4 work with Debian 10, and mE 0.42.1 and mEOS dev20201026.
From previous issues, also tried in Debian12, XFCE4, and installed VLC. VLC showed the video in Media, Network Stream, using the same URL. Installed mE 0.43.1b4 and still the same issue.
Thoughts? Suggestions? Need further info?

MichaIng, 2025-06-17:
Do they work when you remove/disable all other cameras? And do they work if you add them as plain MJPEG network cameras without motion processing?

And with motionEye 0.42.1 on the same hardware, adding the very same multiple cameras the same way works?

It is somewhat expected that there are limitations on how many cameras you can add and watch concurrently via web interface, doubled in case of network cameras which are processed by motion:
- All camera streams need to be send over network to each watching client/browser concurrently.
- For processed network cameras, the input streams come on top.
- If motionEye runs into bandwidth shortage, the camera remains grey in the web UI.

But that should be generally the same with motionEye 0.42.1, since the underlying networking tools and natural bandwidth limitations are the same. Though since it is Python 2 only, some modules will be much older versions, and may behave differently.

starbasessd, 2025-06-17:
> Do they work when you remove/disable all other cameras? 

No, even when I only add one foscam camera by itself.

> And do they work if you add them as plain MJPEG network cameras without motion processing? 

No, will not set up due to the rtsp port being 88 and no other video port on the camera. 

> And with motionEye 0.42.1 on the same hardware, adding the very same multiple cameras the same way works?

Yes, all 4 cameras work simultaneously.
 
> It is somewhat expected that there are limitations on how many cameras you can add and watch concurrently via web interface, doubled in case of network cameras which are processed by motion:
>
>     * [...]

BTW, I was primary support for 0.42.x under Calin, in case you forgot.

MichaIng, 2025-06-17:
> BTW, I was primary support for 0.42.x under Calin, in case you forgot.

Ah right, I didn't notice the name.

Any error messages, and what does debug log say when you visit the web UI?

And is there a way you can make one of foscams streams available publicly for a while?

Ah, and do you use the same latest `motion` version on the Buster system with old mE? There are some compatibility issues with certain settings, but IIRC generally it works. I remember there was some weird behavior or change where motion enforces 640x480 resolution for network cameras or so.

> No, will not set up due to the rtsp port being 88 and no other video port on the camera.

Ah yeah right that seems to not work with RTSP streams. Would be actually a reasonable feature to allow direct streaming from source to browser for RTSP camera streams as well, not MJPEG only. But different topic.

starbasessd, 2025-06-17:
error: Unable to open netcam camera in systemctl log. Will try to grab logs tomorrow when I resume this. 
No, not able to open it to internet. I'm behind 2 firewalls. 
I used the latest motion installed by the install instructions, And also downgraded to 4.5.1 in another clean install in Bullseye. 
I keep my Debian 10 install clean with motion 4.3.2. I can try it on this hardware with Debian 10, 0.42.1 and 4.7.0 tomorrow.

starbasessd, 2025-06-17:
● motioneye.service - motionEye Server
     Loaded: loaded (/etc/systemd/system/motioneye.service; enabled; preset: enabled)
     Active: active (running) since Tue 2025-06-17 18:55:10 EDT; 1min 4s ago
   Main PID: 504 (meyectl)
      Tasks: 10 (limit: 4653)
     Memory: 158.5M
        CPU: 2.137s
     CGroup: /system.slice/motioneye.service
             ├─504 /usr/bin/python3 /usr/local/bin/meyectl startserver -c /etc/motioneye/motioneye.conf
             ├─525 /usr/bin/motion -n -c /etc/motioneye/motion.conf -d 5
             └─566 /usr/bin/python3 /usr/local/bin/meyectl startserver -c /etc/motioneye/motioneye.conf

Jun 17 18:55:40 deb12-meye-c1-2 motion[525]: [1:ml1:Camera1] [ERR] [VID] vid_start: Netcam RTSP failed to open
Jun 17 18:55:50 deb12-meye-c1-2 motion[525]: [1:ml1:Camera1] [WRN] [ALL] mlp_retry: Retrying until successful connection with camera
Jun 17 18:55:50 deb12-meye-c1-2 motion[525]: [1:ml1:Camera1] [ERR] [NET] netcam_rtsp_open_context: norm: Unable to open camera(Camera1): Invalid argument
Jun 17 18:55:50 deb12-meye-c1-2 motion[525]: [1:ml1:Camera1] [ERR] [VID] vid_start: Netcam RTSP failed to open
Jun 17 18:56:00 deb12-meye-c1-2 motion[525]: [1:ml1:Camera1] [WRN] [ALL] mlp_retry: Retrying until successful connection with camera
Jun 17 18:56:00 deb12-meye-c1-2 motion[525]: [1:ml1:Camera1] [ERR] [NET] netcam_rtsp_open_context: norm: Unable to open camera(Camera1): Invalid argument
Jun 17 18:56:00 deb12-meye-c1-2 motion[525]: [1:ml1:Camera1] [ERR] [VID] vid_start: Netcam RTSP failed to open
Jun 17 18:56:10 deb12-meye-c1-2 motion[525]: [1:ml1:Camera1] [WRN] [ALL] mlp_retry: Retrying until successful connection with camera
Jun 17 18:56:10 deb12-meye-c1-2 motion[525]: [1:ml1:Camera1] [ERR] [NET] netcam_rtsp_open_context: norm: Unable to open camera(Camera1): Invalid argument
Jun 17 18:56:10 deb12-meye-c1-2 motion[525]: [1:ml1:Camera1] [ERR] [VID] vid_start: Netcam RTSP failed to open

MichaIng, 2025-06-19:
```
netcam_rtsp_open_context: norm: Unable to open camera(Camera1): Invalid argument
```
Let's check where this is emitted.

MichaIng, 2025-06-19:
https://github.com/Motion-Project/motion/blob/master/src/netcam.cpp#L1827-L1837

So it is `motion` using the FFmpeg library to open the camera device, but it seems to get an invalid argument.

Can you show the camera config, especially the `motion` command call:
```sh
cat /etc/motioneye/camera-1.conf
```
_Should be the correct config path._

HomAlix99, 2025-11-07:
Hi,

I have exact the same issue,
Was this issue solved?

br.

Alex

JamBalaya56562, 2026-11-12:
Marijn's guess above was fixed in 0.44.0 (#3296): motionEye now percent-encodes the username and password it passes to Motion, so passwords with characters like `#` or `@` work. The camera needs to be added again for that. As there was no reply since February, I'm closing this; if the Foscam cameras still stay grey on 0.44.0, please open a new issue with the camera config and Motion's log.

closed 2026-11-12 by JamBalaya56562