motioneye-project/motioneye #3169: E-Mail Notification Error when using multiple Adresses

djkroko, 2025-06-25:
When using multiple E-Mail Adresses in die Notification Settings, sending will fail.

With a single Adress I do get an Error but it still sends the Mail.

```

   ERROR: notification email test failed: {'test1@domain.de,test2@domain.de': (501, b'5.1.3 Invalid address')}
Traceback (most recent call last):
  File "/usr/lib/python3.12/site-packages/motioneye/handlers/config.py", line 681, in test
    sendmail.send_mail(
  File "/usr/lib/python3.12/site-packages/motioneye/sendmail.py", line 74, in send_mail
    conn.sendmail(_from, to, email.as_string())
  File "/usr/lib/python3.12/smtplib.py", line 890, in sendmail
    raise SMTPRecipientsRefused(senderrs)
smtplib.SMTPRecipientsRefused: {'test1@domain.de,test2@domain.de': (501, b'5.1.3 Invalid address')}
   ERROR: mjpg client timed out receiving data for camera 2 on port 9082
   ERROR: notification email test failed: {'test1@domain.de,test2@domain.de': (501, b'5.1.3 Invalid address')}
Traceback (most recent call last):
  File "/usr/lib/python3.12/site-packages/motioneye/handlers/config.py", line 681, in test
    sendmail.send_mail(
  File "/usr/lib/python3.12/site-packages/motioneye/sendmail.py", line 44, in send_mail
    conn = smtplib.SMTP(server, port, timeout=settings.SMTP_TIMEOUT)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/smtplib.py", line 255, in __init__
    (code, msg) = self.connect(host, port)
                  ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/smtplib.py", line 341, in connect
    self.sock = self._get_socket(host, port, self.timeout)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/smtplib.py", line 312, in _get_socket
    return socket.create_connection((host, port), timeout,
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/socket.py", line 841, in create_connection
    for res in getaddrinfo(host, port, 0, SOCK_STREAM):
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/socket.py", line 978, in getaddrinfo
    for res in _socket.getaddrinfo(host, port, family, type, proto, flags):
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
```

MichaIng, 2025-06-26:
Can you try this:
```sh
cd /usr/lib/python3.12/site-packages
curl -sSf 'https://patch-diff.githubusercontent.com/raw/motioneye-project/motioneye/pull/3170.patch' | sudo patch -p1
```
Might be needed to install the `patch` package first. Of course you can also apply the small change manually.

djkroko, 2025-06-27:
I use it in Home Assistant as AddOn. I applied the patch manually and then restarted the AddOn.

Seems like the AddOn works a bit different, after a restart the config.py reverts it to the wrong entry.

MichaIng, 2025-06-29:
It works for me, but somehow the UI shows a white failure message:

[image]

Both mails however arrived, no error in service logs and backend errors are usually indicated in orange test. I'll check where this is coming from.

MichaIng, 2025-06-29:
Okay merged, will be fixed with next beta iteration.

closed 2025-06-29 by MichaIng