motioneye-project/motioneye #3171: motionEye App

SuMaPa, 2025-06-28:
Hi

h**ps://play.google.com/store/apps/details?id=com.jairaj.janglegmail.motioneye

How did you develop this app? Can you use them? Or should you avoid them?

THX

MichaIng, 2025-06-28:
Hi, nope this is not related to the motionEye project itself. Didn't know it until now, interesting. As long motionEye's own web UI is not that mobile friendly, I see the use case.

Here its GitHub repo: https://github.com/JairajJangle/motioneye-android

On the one hand, for a mobile app to make sense, the motionEye instance needs to be accessible from the Internet, and of course the app needs to get credentials. So theoretically it could send credentials somewhere which allows them to login your instance.

On the other hand, the app is open source, and Google does do some security review before releasing apps in their store. While this is no guarantee, and there have and always are cases of malicious apps, the chance in this case is probably rare. But wouldn't hurt it someone checks through the code, tracing variables and methods from the point on where login credentials are entered.

Generally, keep in mind the following: motionEye itself currently is not super secure: it has two hardcoded user names, stores passwords in outdated SHA1 hashes, and has no 2FA or something like that. So exposing it directly to the Internet without any proxy or something with stronger authentication or other client filtering, should probably not be done without a good particular reason, at least if the streams are private/sensitive. In any case, assure to assign strong passwords for both, admin and surveillance user.

Reminds me that we do have a lot of work to do. Contributors who can tackle some hardening and modernization tasks are desperately needed 😅.

shadowkeke, 2025-06-30:
Hello,

The Android app is the latest stable from the store: https://github.com/JairajJangle/motioneye-android
Anroid 13

Server:
motionEye Version 	0.43.1b1
Motion Version 	4.6.0
OS Version 	Linux 6.6.69_2

TBH I don't see any major changelog related to this so I rather don't upgrade if I don't have to.
Besides it's the same with the latest Firefox too:

I'm on Linux no codec packs. 

[image]

MichaIng, 2025-06-30:
Is it the same when accessing motionEye's own web interface? However, likely an Android limitation/topic, with apps just using WebView or Android playback features, let's keep at: #3175

SuMaPa, 2025-07-01:
Hi MichaIng

My Motioneye is running on my ubuntu server and is accessible from outside.
I just installed the app and entered the Motineye login details.
It's working^^

JamBalaya56562, 2026-11-07:
This was answered above: the app is a third-party project and not part of motionEye, so please take questions about it to its own repository: https://github.com/JairajJangle/motioneye-android

Closing, as the app works for you. The playback problem mentioned here continues in #3175.

closed 2026-11-07 by JamBalaya56562