motioneye-project/motioneye #3175: Google Chrome refuses to playback HEVC videos [yes it still is in 2025]

shadowkeke, 2025-06-30:
Hello,

I open this again, as same as last year same as year before. The only advise I get here is why don't I download the videos from motioneye and open it with an external program ... because its not userfriendly thats why.

So here it goes again:

The solution which is suggested in similar threads are always that you need to have Hardware support and it will:

https://support.google.com/chrome/thread/314380188/chrome-video-playback-for-hevc?hl=en

Not true I experiencing the same with my i7 lenovo thinkbook, but even if the hardware does not support decoding it should support it from software.

I found no extensions specifically for this, it's a really old problem I try to resolve.

I using motioneye for camera recordings, it offers different formats to store video files. The goal is to be able to playback videos on both Android phones and PC.

I have literally went through all and the only format Android plays back is HEVC in the motioneye app, all the other's it will not:

    MPEG4
    SWF
    FLV
    Quicktime Mov
    H264
    Matroska
    Matroska OMX

Now quite unfortunately HEVC is the one format which will not play from Chrome.

At playback I get:

Error: Media format unsupported or otherwise unavailable/unsuitable for playing.

I opened a discussion with the devs multiple times they say it's Chrome's fault. The videos of course can be downloaded on both PC and Android regardless what codec I use and opened with a program like VLC but this is unusable.

Any suggestions are welcome!

MichaIng, 2025-06-30:
Not sure what this has to do with motionEye, but interesting topic.

So HEVC plays on Android, but nothing else (which indeed is unexpected), and everything else plays on Chrome (I guess SWF, FLV, and Quicktime do neither), but not HEVC?

And just to be sure, you use motionEye 0.43.y?

What which player do you try on Android, and which version (of both, Android and player)?

Side notes:
- We should remove legacy SWF, FLV, and probably Quicktime (?), if there is any more code involved than asking FFmpeg what it supports and offering its output as is.
- Similarly, if there is any processing of FFmpeg output done, the format names should be clarified: Matroska is a container format, H264 is a codec format (which can be well used with Matroska), and MPEG4 is a conglomerate of different video and audio standards and formats, which includes H264 among others. Probably MP4 is meant, another container format, which is based on Quicktime, and supports as well H264, HEVC and other codec formats. When selecting one of the codec or container formats respectively, probably FFmpeg just chooses some common complementary container or codec format respectively 😄. But actually both are passed to different CLI options, hence unexpected if motionEye mixes those all together.
- I guess we should drop legacy Raspberry Pi firmware formats, including MMAL cameras and the OMX/OpenMAX based stuff, including the `h264_omx` codec, most likely used here when selecting "Matroska OMX" (which at least defines container and codec both).

Pretty well compatible and probably best default for the use case up to 1440p or so, should be H264 codec with Matroska container. So if this is not offered, we should add it.

TiboTake, 2025-06-30:
Hey, install klite codec the mega pack. Now you can all format play on youre pc.

i use this coec pack more than 5 yeahr and it works fine

JamBalaya56562, 2026-11-12:
Which formats a browser plays is up to the browser and the system's decoders; motionEye only serves the file. For playback in any browser, Movies → Movie format "H.264 (.mp4)" is the safe choice, and new cameras use it by default on the dev branch (#3543). Closing, as there is nothing motionEye can change in Chrome's HEVC support.

closed 2026-11-12 by JamBalaya56562