motioneye-project/motioneye #3178: Email password in notification cannot contain the dollar '$' symbol

gianlucagiacometti, 2025-07-13:
An error occurs when the email password contains a '$' symbol

[image]

If I remove the '$' everything works fine.

gianlucagiacometti, 2026-02-05:
Well, I tried with the same password that didn't work at that time and now it woks. I also tried different passwords containing the "$" symbol and they all work.
I set an automatic nightly update of motioneye, so I suppose that some changes had been done in the past that solved the problem.

MichaIng, 2026-02-05:
Weird, AFAIK nothing changed around handling email passwords. However, great that it works now. Then I'll close the issue.

closed 2026-02-05 by MichaIng