motioneye-project/motioneye #3179: MotionEye with Anker PowerConf C200 on Raspberry5 (bookworm) stuck at 640x480 resolution

BrianEmberling, 2025-07-13:
I've changed the resolution in motionEye to 1280x720 (and others) but the output video is always 640x480 (yes, I hit "apply").
I've tried this webcam on two different RPI's, and they are both stuck at 640x480. 
I've tried a different webcam (Logictech 920) and that works fine at different resolutions.

MotionEye version 0.43.1b2, Motion version 4.7.0, OS version Debian 12

The /var/logs/motioneye/motion.log has:  (full motion.log is linked at the end)
    1:ml1:Camera1] [WRN] [VID] v4l2_pixfmt_adj: **Adjusting resolution from 1280x720 to 640x480.**

The camera shows it is capable of 1280x720 via v4l2-ctl:
% **v4l2-ctl -d /dev/video0 --list-formats-ext**
ioctl: VIDIOC_ENUM_FMT
	Type: Video Capture

	[0]: 'MJPG' (Motion-JPEG, compressed)
		Size: Discrete 2560x1440
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 1920x1080
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 1280x720
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 640x480
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 640x360
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 320x240
			Interval: Discrete 0.033s (30.000 fps)
	[1]: 'YUYV' (YUYV 4:2:2)
		Size: Discrete 640x480
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 640x360
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 320x240
			Interval: Discrete 0.033s (30.000 fps)
	[2]: 'H264' (H.264, compressed)
		Size: Discrete 2560x1440
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 1920x1080
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 1280x720
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 640x480
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 640x360
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 320x240
			Interval: Discrete 0.033s (30.000 fps)

[image]

[image]

**motion.log:**

[motion.log](https://github.com/user-attachments/files/21206094/motion.log)

BrianEmberling, 2025-07-13:
I suppose the answer is in the log and camera capabilities list: it only supports high resolution for H.264 and MJPEG output, not for any YUV format.
Does anyone know how to get motion/motioneye to support these?

Marijn0, 2026-10-31:
I believe you can use the `palette` option in `video_params` to make Motion use MJPEG instead of YUYV: 
https://motion-project.github.io/4.6.0/motion_config.html#video_params

So I would try adding `palette=8` at the end of the existing `video_params` line in the camera config and then restart Motion/motionEye. Note that motionEye rebuilds `video_params` from the camera controls whenever you save the camera settings in the UI, so the `palette=8` will be removed again after that.

JamBalaya56562, 2026-11-15:
As you found, the camera offers 1280x720 and up only in MJPEG and H.264, and motion picked YUYV, which stops at 640x480. `palette=8` in `video_params`, as @Marijn0 suggested, makes motion use MJPEG. Saving the camera settings used to drop it again; #3551 fixes that, so from v0.45.0 you can add `video_params palette=8` under Extra Motion Options and it stays. Closing; please open a new issue if the resolution is still stuck with it.

closed 2026-11-15 by JamBalaya56562