motioneye-project/motioneye #3181: Question - Docker compose

ecsufio, 2025-07-17:
Good evening everyone,
Simple beginner's question, I want to install MotionEye via docker compose, with a specific user (in my example, UID/GID 1000:1000). Can I add the following lines to the docker-compose script?

    environment:
      - PUID=1000
      - PGID=1000

Or will this not work?

Thanks for your help!

ecsufio, 2025-07-17:
Thank you for your quick reply.
If I understand correctly, here it's better to create an image with the dockerfile rather than using the docker-compose script, right?

MichaIng, 2025-07-17:
Ah yeah right, that only works when generating a new image. motionEye (and `motion`) within the container runs as `motion` user, mapped to UID/GID 0 within the container, by default, i.e. with the images provided via container registry. 

I see `PUID`/`PGID` is sort of a convention to make this dynamic. Looks like a reasonable feature we should implement, and replace the `RUN_UID`/`RUN_GID`.

zagrim, 2025-07-19:
To start, I must confess I'm not an expert on Docker at all, but I've done some research on container security due to running stuff in Kubernetes at work, and I have an interest in practical infosec in general.

>But the access of the container is limited to the container image and volumes, so security implications are naturally limited, being one of the reasons for using containers anyway.
>
>Optional UID/GID is about what the related software daemon within the container runs as. This is not really a security feature, but for convenience, since this is the user which creates config/data in the volumes

@MichaIng I agree with what you said about the optional UID/GID being more for convenience for MotionEye, and it is true that running things in container adds one level of isolation between the host system and the application/daemon run inside the container.

However, in general it is a commonly recommended best practice these days (like in [OWASP Docker security cheat sheet](https://cheatsheetseries.owasp.org/cheatsheets/Docker_Security_Cheat_Sheet.html#rule-2-set-a-user)) that one should not run stuff as root (UID 0 / GID 0, it doesn't really matter if it's called "motion" or "root") since it also has consequences to what level of access the process **within** the container has: non-root users cannot modify system files, install packages etc. Non-root users also cannot access root owned files in the host filesystem (or create files owned by host root user) even if the container had been given a volume mount that was not thought through. Also, in case of container escape, if the attacker had root access inside the container, they would immediately gain root access also to the host system.

The importance of all this depends of course on the threat model. For a home user who only allows access to MotionEye from their local LAN/Wifi, and trusts their fellow users in the network, it's probably not a big deal if stuff is run as UID 0 in the container. But things change if one allows access to MotionEye from the internet, uses it for business purposes, tweaks the configs to adjust to some custom needs, mounts host file system carelessly etc and/or just isn't techically savvy enough to understand what they are doing when following some random advice found on the Internet.

In a simple case where no mounted volumes are needed and you don't have dependencies within the image that expect some specific user, running as a custom non-root user in container is a piece of cake, but indeed it gets a bit more complex when you have dependencies and need to access stuff external to the container, or when you provide your images for others to run... 😅

ecsufio, 2025-07-19:
Hello everyone,
Thank you, @MichaIng, for your quick replies!
I'm glad my simple question has helped to move your project forward!
I'm generally interested in security issues. I don't have any particular programming skills, but please let me know if there's anything I can do to help with this project.

In any case, I'm discovering MotionEye (running as root :wink: ) and so far everything is going as expected. Thank you!

MichaIng, 2026-10-29:
#3484 instead allows to use `docker run --user`, resp. the `user:` key in Docker Compose YAML. The README and example YAMLs has been updated to cover that: https://github.com/motioneye-project/motioneye/blob/dev/docker/README.md#running-as-a-non-root-user

closed 2026-10-29 by MichaIng