motioneye-project/motioneye #3187: Separate viewing settings for each viewing end-point (browser / Android app etc.)

xj25vm, 2025-08-08:
I am not familiar with the MotionEye internals and not even sure if this is feasible. I have a number of instances of MotionEye setup to which several users connect to - some from web browsers, some from the Android app. As things are at the moment, if one of the users changes a viewing setting (number of columns, resolution dimmer, fps dimmer), all other users connected to the same MotionEye instance are affected. Would it be possible for these settings to 'belong' to each end point separately? For example, I've had users 'experimenting' with the controls in the interface and freezing everyone's cameras - as they turned the fps dimmer all the way down to zero - for example.

Another reason separate settings would be useful is that, for example, if one end point is remote on a phone app, it makes sense to turn resolution/fps all the way down to cope with a slow connection - but someone viewing it on the local network might want high quality settings as the bandwidth is not an issue. I think at the moment the only way this is achievable is by running several instances of MotionEye connected to same cameras?

Or maybe this would require support for multiple user accounts, each with their own settings?

JamBalaya56562, 2026-11-13:
motionEye 0.45.0b1 keeps these per browser (#3502): the layout columns and rows and the frame rate and resolution dimmers are saved in each browser, so one viewer turning them down no longer affects the others.

closed 2026-11-13 by JamBalaya56562

SebastianArcus, 2026-11-13:
Thank you very much for adding this feature - much appreciated!